nerves-livebook/nerves_livebook #19: Figure out a nice way to structure a small Livebook program

fhunleth, 2021-05-07:
The idea is it should be possible to write a short program in Elixir that runs indefinitely in Nerves Livebook. For an Arduino, this would be filling in `setup()` and `loop()` functions. I am not sure how this best maps to Elixir, but I think that it would be pretty neat. 

If anyone has ideas, feel free to add to this issue or link to a proof-of-concept.

PhillippOhlandt, 2022-07-09:
Maybe add an empty DynamicSupervisor to the main supervision tree. Since in Nerves Livebook, everything is one instance, it would be really easy to write some code (supervisor, genserver, etc.) in a livebook and start that under the dynamic supervisor, where it then keeps running until a restart or manual stop.

a-maze-d, 2023-12-28:
I love the idea. You might want to get more fancy than that and allow it to survive a restart.

Maybe register a (or several)  livebook(s) that gets executed after startup. This would be very powerful especially in combination with the above mentioned DynamicSupervisor.

Only thing to think about would be on how to have a way to avoid the automatic startup (for the case that there is some nasty bug in the application). Otherwise the user might end up in an endless loop.