nerves-livebook/nerves_livebook #36: Add picam example

fhunleth, 2021-05-21:
It seems like it should be possible to stream video to a Livebook with picam. 

Fun features would be:

* Changing the filter
* Changing sizes
* Text overlays

Picam can be set to stream a motion JPEG so a web page with an image tag is sufficient to show the results. I'm not sure how to put the image tag in the Livebook page so that it works, though. It might be that a separate browser window is needed where the user can hit reload for they start and stop whatever GenServer is streaming the image to the image tag.

PhillippOhlandt, 2022-07-09:
The poor mans version using pure livebook is this:

```
Picam.Camera.start_link()
Picam.set_size(720, 0)
```

```
frame = Kino.Frame.new()
```

```
for i <- 1..1000 do
  Kino.Frame.render(frame, Kino.Image.new(Picam.next_frame(), "image/png"))
  Process.sleep(50)
end
```

Each code block would be a code cell. The loop could be done using something else but I think it's a good idea to not let it run forever without a way to stop it.

fhunleth, 2023-07-26:
Closing since the Raspberry Pi Nerves systems are moving away from MMAL-based camera support. This will completely break `picam`. It would be better to get `libcamera` better supported in Elixir.

closed 2023-07-26 by fhunleth