nerves-livebook/nerves_livebook #156: nerves_livebook_rpi4.fw(v0.5.0) does not show a graph

TORIFUKUKaiou, 2022-01-24:
## Environment

- nerves_livebook_rpi4.fw
- v0.5.0

## Current behavior

1. I download and burn into microSD: `fwup nerves_livebook_rpi4.fw`
1. Run Raspberry Pi 4 with microSD burned nerves_livebook_rpi4.fw
1. http://localhost:4000
1. EXPLORE > See all > Plotting with VegaLite
1. Evaluate, Evaluate
1. Evaluate Basic concepts

It does not show a graph.

[image]

## Expected behavior

It shows a graph.

- Livebook 0.5.1 works well.
- Nerves Livebook v0.4.2 works well.

TORIFUKUKaiou, 2022-01-24:
```
ssh nerves.local

iex> RingLogger.attach
```

I see below.

```
16:56:01.578 [info]  GET /sessions/d66liv6we4kofxturu2fjz4kis62dffyl2bjxrmh/assets/aonl63aeaytrz5n5ufznu7icvy/main.js
                                  
16:56:01.579 [debug] Processing with LivebookWeb.SessionController.show_asset/2
  Parameters: %{"file_parts" => ["main.js"], "hash" => "aonl63aeaytrz5n5ufznu7icvy", "id" => "d66liv6we4kofxturu2fjz4kis62dffyl2bjxrmh"}
  Pipelines: [:js_output_assets]
                                  
16:56:01.584 [error] GenServer {Livebook.UniqueTask.Registry, "aonl63aeaytrz5n5ufznu7icvy"} terminating
** (MatchError) no match of right hand side value: {:error, "no such file or directory"}
    (livebook 0.5.1) lib/livebook/session.ex:170: anonymous fn/3 in Livebook.Session.fetch_assets/2
    (livebook 0.5.1) lib/livebook/unique_task.ex:19: Livebook.UniqueTask.Task.handle_continue/2
    (stdlib 3.17) gen_server.erl:695: :gen_server.try_dispatch/4
    (stdlib 3.17) gen_server.erl:437: :gen_server.loop/7
    (stdlib 3.17) proc_lib.erl:226: :proc_lib.init_p_do_apply/3
Last message: {:continue, #Function<5.38989450/0 in Livebook.Session.fetch_assets/2>}
State: nil
```

Hope this helps.

mnishiguchi, 2022-01-24:
I checked it with my `rpi0` and I saw the same result: 
- graphs not displayed
- the same error @TORIFUKUKaiou reported occurred
[image]

fhunleth, 2022-01-25:
Thanks for the report. Have either of you looked into what file it's trying to read?

mnishiguchi, 2022-01-25:
As I logged the `local_assets_path` value right before the error location, it looks like: 

```
"/tmp/livebook/assets/aonl63aeaytrz5n5ufznu7icvy"
```

- https://github.com/livebook-dev/livebook/blob/5110f85e7389c2b0d7c76183ee0b0c2cb5df6561/lib/livebook/session.ex#L168

```elixir
  defp local_assets_path(hash) do
    Path.join([livebook_tmp_path(), "assets", encode_path_component(hash)])
  end
```

jonatanklosko, 2022-01-25:
We introduced user-defined JS widgets and migrated the Vega-Lite plot to use that. Each widget module has assets .tar.gz bundle generated at compile time and it is put in the `:kino` priv directory (so it gets bundled into the release):

[image]

When a widget is actually rendered, Livebook copies this bundle into a cache directory (because theoretically the runtime could be on another machine). @fhunleth I guess the problem is that the file system is readonly, so Livebook fails to write to `/tmp/livebook/assets/aonl63aeaytrz5n5ufznu7icv`, and consequently it doesn't find the file there later. Does that make sense?

mnishiguchi, 2022-01-25:
Ahhhh, @fhunleth My `archive_path` points to my host machine. It might be the cause.

```
iex(livebook@nerves-mn02.local)> Kino.VegaLite.__assets_info__
%{
  archive_path: "/Users/mnishiguchi/src/nerves_livebook/_build/rpi0_dev/lib/kino/priv/assets/Kino_VegaLite/aonl63aeaytrz5n5ufznu7icvy.tar.gz",
  hash: "aonl63aeaytrz5n5ufznu7icvy",
  js_path: "main.js"
}
```

jonatanklosko, 2022-01-25:
Oh, that makes sense. Looking again at the stacktrace I've just realised that the problem is actually in reading the `:archive_path` itself and not reading from the cache location later on.

So the issue is that we compile Kino when building the release, so the absolute path is specific to the machine where it was built.

jonatanklosko, 2022-01-25:
Could you try pointing Kino in `nerves_livebook` to:

```elixir
{:kino, github: "livebook-dev/kino", branch: "jk-archive-path"}
```

and checking if it helps?

mnishiguchi, 2022-01-25:
@jonatanklosko It worked 🎉 
- all graphs are rendered
- no error occurred
- my `archive_path` points to a correct one

```
iex(livebook@nerves-mn02.local)> Kino.VegaLite.__assets_info__
%{
  archive_path: "/srv/erlang/lib/kino-0.5.0/priv/assets/Kino_VegaLite/aonl63aeaytrz5n5ufznu7icvy.tar.gz",
  hash: "aonl63aeaytrz5n5ufznu7icvy",
  js_path: "main.js"
}
```

fhunleth, 2022-01-25:
@mnishiguchi, @jonatanklosko, and @TORIFUKUKaiou Thank you all for finding this, debugging and fixing! I'll keep watching and make a new `nerves_livebook` release when the time is right.

jonatanklosko, 2022-01-25:
@mnishiguchi fantastic, thanks for verifying!

@fhunleth if you pointed Kino dependency to GitHub, would `Mix.install({:kino, "~> 0.5.0})` work? Otherwise we will make a new release :)

jonatanklosko, 2022-01-25:
Actually, will ship Kino `0.5.1` shortly either way :)

jonatanklosko, 2022-01-25:
Done :ship:

fhunleth, 2022-01-25:
I just verified that the plotting with VegaLite notebook works now! Thanks everyone! Release coming shortly.

closed 2022-01-25 by fhunleth

TORIFUKUKaiou, 2022-01-25:
Thank you all a lot!

commit 2d02b700 (2022-01-25): Bump kino for asset path fix

Fixes #156