nerves-livebook/nerves_livebook #192: List available dependencies in the dependency search

jonatanklosko, 2022-04-01:
This needs to wait for the next Livebook release. As of https://github.com/livebook-dev/livebook/pull/1081 we have a dependency search. For the default runtime it searches on Hex. For the embedded runtime there is a configuration option that allows for listing predefined dependencies, it works like this:

```elixir
config :livebook, Livebook.Runtime.Embedded,
  load_dependency_entries: {NervesLivebook.Dependencies, :dependency_entries, []}

defmodule NervesLivebook.Dependencies do
  def dependency_entries() do
    # Return dependencies in this format:
    [
      %{
        dependency: {:kino, "~> 0.5.2"},
        description: "Interactive widgets for Livebook",
        name: "kino",
        url: "https://hex.pm/packages/kino",
        version: "0.5.2"
      }
    ]
  end
end
```

jonatanklosko, 2022-04-01:
Also, just a heads up, the same PR made some changes to runtimes, so we will need to change the config

```diff
config :livebook,
-  default_runtime: {Livebook.Runtime.Embedded, []}
+  default_runtime: Livebook.Runtime.Embedded.new()
```

fhunleth, 2022-04-03:
Cool. Thanks for the heads up.

jonatanklosko, 2022-04-04:
The embedded runtime won't be available by default, so we will also need to reconfigure like this:

```elixir
config :livebook, :runtime_modules, [Livebook.Runtime.Embedded, Livebook.Runtime.Attached]
```

(ref https://github.com/livebook-dev/livebook/issues/1082)

fhunleth, 2022-05-04:
I updated to Livebook 0.6.0 in #207. The dependency search results still need to be dynamically generated, though.

commit f5884d7f (2022-05-07): Return built-in packages in the dependency search

Fixes #192
commit 516b571c (2022-05-07): Return built-in packages in the dependency search

Fixes #192
commit c4316e10 (2022-05-07): Return built-in packages in the dependency search

Fixes #192
commit 03e87ec7 (2022-05-07): Return built-in packages in the dependency search

Fixes #192
closed 2022-05-07 by fhunleth