nerves-livebook/nerves_livebook #198: Building Source Code does not boot Livebook

KosmonautX, 2022-04-13:
Building the firmware directly from the [source](https://github.com/livebook-dev/nerves_livebook/archive/refs/tags/v0.5.7.zip) following the steps below results in the RPI 4 to continue blinking but not connecting to the WIFI (seen from network scan)

```
# Set the MIX_TARGET to the desired platform (rpi0, bbb, rpi3, etc.)
$ export MIX_TARGET=rpi4
$ mix deps.get
$ mix firmware
$ MIX_TARGET=rpi4 NERVES_WIFI_SSID='$WIFI' NERVES_WIFI_PASSPHRASE='$PASS' mix burn
I tried option 2 as well after booting the compiled install as well but it disconnects from the ip address and does not run the process at port 80
# Option 2: Upload to an existing Nerves Livebook device
$ mix firmware.gen.script
$ ./upload.sh livebook@172.20.10.10
```

The [compiled release] (https://github.com/livebook-dev/nerves_livebook/releases/download/v0.5.7/nerves_livebook_rpi4.fw ) works great but I can't modify the mix.exs to include a phoenix client library I need to directly communicate to websockets

fhunleth, 2022-04-14:
That looks like it should work. I'm out of town and don't have a Pi4 to test with. Since I do this regularly, I suspect that it would work for me and there's something else going that's not immediately obvious to me.

One question, does this work without the phoenix client library added to the dependency list? Like could it be related to the dependency or one of its dependencies? This happens every once in a while and usually it's not a hard fix for me if you tell me the library.

KosmonautX, 2022-04-14:
Indeed I had suspected the same and removed from the mix.exs file when it was initially not working
```
{:phoenix_client, "~> 0.3", , targets: @rpi_targets},
```

I  haddecided to eliminate all variables and try a clean build from source and see where I was going wrong from there when I encountered this issue. I was unsure how to go about debugging and arriving at the core of the issue when I couldn't connect to the RPI over the wifi network and then tried option 2 before I opened the issue here.

I am thinking of trying out the process again with another RPI I have in school after this weekend and see where it leads me.