nerves-livebook/nerves_livebook #226: Links from Nerves "explore" notebook don't work

fhunleth, 2022-06-25:
I haven't looked into this, yet, but if you click on the Nerves explore notebook and then try any of the links in it, they won't work. The links are all relative ones and it looks like there either needs to be a way to specify a base path or they need to be made into absolute ones.

jonatanklosko, 2022-06-25:
The explore notebooks are opened from memory (stored in module code at compilation time), so they don't really have a file. The other notebooks need to be include in the configuration, they can have `details: nil` to not appear in the UI themselves. Then in the main notebook can have links like `/explore/notebooks/:slug`. For example, in Welcome to Livebook we have `[Elixir and Livebook](/explore/notebooks/elixir-and-livebook)`.

fhunleth, 2022-06-25:
Cool. Thanks!

fhunleth, 2022-10-17:
Fixed in #289

closed 2022-10-17 by fhunleth