nerves-livebook/nerves_livebook #236: Let the Livebook update banner show Nerves Livebook updates instead of regular Livebook updates

PhillippOhlandt, 2022-07-09:
When the regular Livebook got a new release, the update banner in Nerves Livebook gets triggered. 

Since Livebook and Nerves Livebook don't have matching version numbers and release dates, this is misleading and confusing. 

The update check happens in this GenServer:
https://github.com/livebook-dev/livebook/blob/main/lib/livebook/update_check.ex

A simple solution would be to make the github release url and the application name configurable using application config so that Nerves Livebook can set its own values.

@fhunleth pointed out that it would also be nice if the user can interact with the update banner to trigger a firmware update.

a-maze-d, 2024-08-14:
I have raised both an [idea](https://github.com/livebook-dev/livebook/discussions/2742) and a [PR](https://github.com/livebook-dev/livebook/pull/2743) against livebook. This should allow to make this configurable.

It does not solve the second part of the issue, to allow interaction with the banner, but it's at least a start :-) We can move the interaction aspect to a separate issue (should we be able to resolve the first aspect)

a-maze-d, 2024-08-15:
Good news, the PR has been approved and been merged in. Not sure it will find it's way into livebook v0.14 (probably to close to releasing), but at least we should be able to solve the problem as from v0.15 :-)

fhunleth, 2024-08-15:
Wow, thank you!

a-maze-d, 2024-08-21:
My changes made it onto the 0.14 branch. So next upgrade should allow us to make the changes

fhunleth, 2024-08-22:
@a-maze-d Congratulations on getting that PR into Livebook! This is going to be a really nice improvement.

a-maze-d, 2024-09-15:
Yes, the idea was great, and I wanted to raise a PR against nerves_livebook, but it turned out that the app_version (in livebook) is hardcoded and can't be overwritten :-(

I have now raised a [second PR](https://github.com/livebook-dev/livebook/pull/2780) against livebook to get that fixed too :smile:

a-maze-d, 2024-09-20:
Second PR went in... so now we need a bit more and I can raise a PR once the new version is out. I have a hacked version and it's looking good :-)

a-maze-d, 2024-10-19:
The version number is now correctly detected :-) However, the message is still very confusing, because the message is hard-coded [here](https://github.com/livebook-dev/livebook/blob/main/lib/livebook_web/live/home_live.ex). Thus, it will still talk about [livebook.dev](https://livebook.dev/) and not about nerves_livebook.

It is possible to set an instruction URL, but that's does not help. I don't think we can overwrite the [update notification](https://github.com/livebook-dev/livebook/blob/main/lib/livebook_web/live/home_live.ex#L58)

fhunleth, 2024-10-20:
@a-maze-d Thank you keeping up with this. Even though it's not perfect (the message), it seems better than what we currently have. Would you like to do the honors of sending the PR that adds the `:github_release_info`? We can still keep this issue open to track getting a fix to the hard-coded message.

a-maze-d, 2024-10-20:
Will do, but might take a couple of days

fhunleth, 2024-10-20:
No rush. I'm trying to finish up Nerves systems updates before making a new Nerves Livebook release.

a-maze-d, 2024-10-23:
[PR 580](https://github.com/nerves-livebook/nerves_livebook/pull/580) created :-)