nerves-livebook/nerves_livebook #485: Rpi zero w 2 does not boot with recent releases

arrowcircle, 2024-01-02:
Hey! I found, that recent releases does not work with Rapsberry pi zero w 2.
Latest version, that works is v0.10.1 with rpi3a target.
v0.11.1 and v0.12.0 does not boot with black screen and blinking cursor.
I tried both rpi3a and rpi0_2 targets.

Not sure what changed, but it does not work :(

fhunleth, 2024-01-02:
Ugh. Thanks for the report. That's frustratingly the Raspberry Pi that I recently broke and can't easily test. 

However, both of those systems are headless, so I wouldn't expect anything helpful on the HDMI output. Just to double check what you're seeing, when you connect those devices via a USB cable to your computer, do you see a USB device get found? Can you ping it?

Also, do you have a UART cable? There might be an error message on the serial pins on the GPIO header.

fhunleth, 2024-01-02:
Update: I know the problem. I'm working on a fix.

fhunleth, 2024-01-02:
I updated `main` with a partial fix. The updated version of Linux used in the newer releases changed the name of the LED and that is enough to cause one of the OTP applications in Nerves Livebook to not start.

There's unfortunately a second issue which is that somehow the data partition gets corrupted so badly that Livebook can't write to it and fails to start. This is amazing, so I want to spend time debugging it to understand it better. The workaround which worked once for me is to change `/data/livebook/` to `/data` at https://github.com/nerves-livebook/nerves_livebook/blob/10a83e3e040237dbb7d14b7511c1769bbb200d74/config/runtime.exs#L13.

fhunleth, 2024-01-02:
Update: I don't think the workaround does anything. It appears that this just sometimes works and sometimes doesn't or I have a flaky MicroSD and haven't figured it out yet.

I've been testing on an RPi Zero. I believe that you should be able to ssh into the RPi Zero 2 over USB and run `Application.ensure_all_started(:nerves_livebook)` to see an error.

arrowcircle, 2024-01-03:
Hey! Thanks for the answers, I tried one cable and cant ping via USB. Dont know if it's a cable problem or rpi don't boot.
Before that I was using wifi and recent versions just blink after power on.
So any ideas how to fix this?

fhunleth, 2024-01-03:
Are you using Mac, Windows, or Linux on your computer?

Are you able to build and try `main`? I think that you're running into the LED name change issue and maybe not the second issue that I started debugging. 

Fwiw, the way to debug is to attach a USB->UART cable to the UART pins on the GPIO header. There will be an iex prompt and you can run `RingLogger.next` to get log information or try starting OTP applications again to get a longer error message.

bissonex, 2024-01-09:
Hi, I have the same issue with a Rpi Zero W (nerves_livebook 0.12.0 (13e8db13-cab2-55e4-7c13-d0e1ad542512) arm rpi0). Here is the requested serial output from `RingLogger.next`:

```
18:51:54.088 [warn] usb_phy_generic phy: supply vcc not found, using dummy regulator
18:51:54.111 [warn] mmc0: host does not support reading read-only switch, assuming write-enable
18:51:54.114 [warn] Kernel memory protection not selected by kernel config.
18:51:54.126 [error] EXT4-fs (mmcblk0p3): VFS: Can't find ext4 filesystem
18:51:54.127 [warn] erlinit: Cannot mount /dev/mmcblk0p3 at /root: Invalid argument
18:51:54.128 [warn] erlinit: Unable to create seed directory
18:51:54.201 [warn] Ignoring non-zero exit status from mount ["-t", "ext4", "-o", "rw", "/dev/mmcblk0p3", "/root"]
18:51:54.238 [warn] Formatting application partition. If this hangs, it could be waiting on the urandom pool to be initialized
18:51:54.516 [warn] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
18:51:54.517 [warn] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
18:51:54.556 [warn] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
18:51:54.906 [warn] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
18:51:54.956 [warn] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
18:51:55.036 [warn] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
18:51:55.946 [warn] dwc2 20980000.usb: supply vusb_d not found, using dummy regulator
18:51:55.947 [warn] dwc2 20980000.usb: supply vusb_a not found, using dummy regulator
18:51:55.947 [warn] dwc2 20980000.usb: Configuration mismatch. dr_mode forced to device
18:51:56.062 [warn] using random self ethernet address
18:51:56.063 [warn] using random host ethernet address
18:51:56.074 [warn] using random self ethernet address
18:51:56.074 [warn] using random host ethernet address
18:52:34.529 [error] [NervesSSH] :ssd.daemon failed: {:error, ~c"No host key available"}
18:52:34.631 [warn] RouteManager: new set_connection_status usb0 -> :disconnected (Elixir.VintageNet.Interface.handle_event/4([file: ~c"lib/vintage_net/interface.ex", line: 343]))
18:52:34.643 [warn] RouteManager: new set_connection_status wlan0 -> :disconnected (Elixir.VintageNet.Interface.handle_event/4([file: ~c"lib/vintage_net/interface.ex", line: 343]))
18:52:34.651 [warn] RouteManager: new set_connection_status usb0 -> :disconnected (Elixir.VintageNet.Interface.start_configuring/3([file: ~c"lib/vintage_net/interface.ex", line: 669]))
18:52:34.710 [warn] RouteManager: new set_connection_status wlan0 -> :disconnected (Elixir.VintageNet.Interface.start_configuring/3([file: ~c"lib/vintage_net/interface.ex", line: 669]))
18:52:34.893 [warn] RouteManager: new set_connection_status usb0 -> :disconnected (ifdown)
18:52:35.032 [error] [NervesSSH] terminating with reason: {:ssh_daemon_error, {:error, ~c"No host key available"}}
18:52:35.036 [error] GenServer NervesSSH terminating
** (stop) {:ssh_daemon_error, {:error, ~c"No host key available"}}
Last message: {:continue, :start_daemon}
18:52:47.180 [warn] version check failed, failed to make a request, reason: {:failed_connect, [{:to_address, {~c"api.github.com", 443}}, {:inet, [:inet], :nxdomain}]}
30 out of 30 new messages shown.
```

fhunleth, 2024-01-09:
@bissonex Thank you. Could you run `NervesMOTD.print`?

bissonex, 2024-01-10:
Sure, 
```
nerves_livebook 0.12.0 (13e8db13-cab2-55e4-7c13-d0e1ad542512) arm rpi0
  Serial       : 0860
  Uptime       : 1 hours, 57 minutes and 2 seconds
  Clock        : 2023-12-17 20:48:26 UTC (unsynchronized)
  Temperature  : 30.9°C

  Firmware     : Valid (A)               Applications : 110 started (delux, nerves_livebook not started)
  Memory usage : 83 MB (26%)             Part usage   : 0 MB (0%)
  Hostname     : nerves-0860             Load average : 0.69 0.62 0.58

  usb0         : 172.31.230.65/30

Nerves CLI help: https://hexdocs.pm/nerves/iex-with-nerves.html
```

fhunleth, 2024-01-21:
Sorry for the long delay. I just posted https://github.com/nerves-livebook/nerves_livebook/releases/tag/v0.12.1. It has the RPi Zero 2W fix and a few others.

arrowcircle, 2024-02-10:
Hey, thanks for possible fixes, but my rpi2 still does not work with recent releases. I ordered proper microusb data cable to investigate it.

fhunleth, 2024-02-10:
@arrowcircle Ugh. That's frustrating. I'll be interested to see what you find. Just to be clear, you wrote "rpi2". Do you have a Raspberry Pi Zero 2W? Download the `rpi0_2` version for that. If you have a Raspberry Pi 2 Model B, then `rpi2` is the right one. The Raspberry Pi 2 is really old and the MicroUSB port is only for power, so it won't be helpful.

arrowcircle, 2024-02-10:
Its rpizero w2, I tried both rpi0_2 and rpi3a. Will post more info when I'll have it

arrowcircle, 2024-02-17:
I've got the cable, here is result of RingLogger.next

```
00:29:07.768 [warn] usb_phy_generic phy: supply vcc not found, using dummy regulator
00:29:07.772 [warn] mmc0: host does not support reading read-only switch, assuming write-enable
00:29:07.784 [warn] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
00:29:07.794 [warn] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
00:29:15.008 [warn] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
00:29:15.071 [warn] RouteManager: new set_connection_status usb0 -> :disconnected (Elixir.VintageNet.Interface.handle_event/4([file: ~c"lib/vintage_net/interface.ex", line: 343]))
00:29:15.074 [warn] RouteManager: new set_connection_status wlan0 -> :disconnected (Elixir.VintageNet.Interface.handle_event/4([file: ~c"lib/vintage_net/interface.ex", line: 343]))
00:29:15.348 [warn] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
00:29:15.378 [warn] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
00:29:15.428 [warn] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
00:29:15.780 [warn] RouteManager: new set_connection_status wlan0 -> :disconnected (Elixir.VintageNet.Interface.start_configuring/3([file: ~c"lib/vintage_net/interface.ex", line: 669]))
00:29:16.167 [warn] dwc2 3f980000.usb: supply vusb_d not found, using dummy regulator
00:29:16.168 [warn] dwc2 3f980000.usb: supply vusb_a not found, using dummy regulator
00:29:16.168 [warn] dwc2 3f980000.usb: Configuration mismatch. dr_mode forced to device
00:29:16.271 [warn] RouteManager: new set_connection_status usb0 -> :disconnected (Elixir.VintageNet.Interface.start_configuring/3([file: ~c"lib/vintage_net/interface.ex", line: 669]))
00:29:16.278 [warn] using random self ethernet address
00:29:16.278 [warn] using random host ethernet address
00:29:16.278 [warn] using random self ethernet address
00:29:16.279 [warn] using random host ethernet address
00:29:16.311 [warn] RouteManager: new set_connection_status usb0 -> :disconnected (ifdown)
00:29:23.552 [warn] version check failed, failed to make a request, reason: {:failed_connect, [{:to_address, {~c"api.github.com", 443}}, {:inet, [:inet], :nxdomain}]}
21 out of 21 new messages shown.
```

```
VintageNet.info
VintageNet 0.13.5

All interfaces:       ["lo", "usb0", "wlan0"]
Available interfaces: []

Interface usb0
  Type: VintageNetDirect
  Present: true
  State: :configured (0:07:18)
  Connection: :lan (0:07:18)
  Addresses: fe80::6cdc:deff:feb2:585c/64, 172.31.244.25/30
  MAC Address: "6e:dc:de:b2:58:5c"
  Configuration:
    %{type: VintageNetDirect, vintage_net_direct: %{}}

Interface wlan0
  Type: VintageNetWiFi
  Present: true
  State: :configured (0:07:19)
  Connection: :disconnected (0:07:19)
  MAC Address: "e4:5f:01:76:ff:34"
  Configuration:
    %{
      type: VintageNetWiFi,
      vintage_net_wifi: %{networks: []},
      ipv4: %{method: :disabled}
    }

:ok
```

Also, when I connect via regular microusb power port, it dos not work and blinks slowly.
If I connect power via data port, it boots, but wifi is disconneected.

fhunleth, 2024-02-18:
This sounds promising. When you're connected via USB, try running `VintageNetWiFi.quick_configure("your_wifi_ssid", "password")`. Then it will connect to your WiFi network from then on whether or not you're connected via USB.

arrowcircle, 2024-02-19:
```elixir
iex(livebook@nerves-32dc.local)2> VintageNet.info
VintageNet 0.13.5

All interfaces:       ["lo", "usb0", "wlan0"]
Available interfaces: []

Interface usb0
  Type: VintageNetDirect
  Present: true
  State: :configured (43.5 s)
  Connection: :lan (43.1 s)
  Addresses: fe80::b058:2bff:fe33:b503/64, 172.31.244.25/30
  MAC Address: "b2:58:2b:33:b5:03"
  Configuration:
    %{type: VintageNetDirect, vintage_net_direct: %{}}

Interface wlan0
  Type: VintageNetWiFi
  Present: true
  State: :configured (44.0 s)
  Connection: :disconnected (44.7 s)
  MAC Address: "e4:5f:01:76:ff:34"
  Configuration:
    %{
      type: VintageNetWiFi,
      vintage_net_wifi: %{
        networks: [
          %{
            mode: :infrastructure,
            psk: "....",
            sae_password: "....",
            ieee80211w: 2,
            key_mgmt: :wpa_psk,
            ssid: "Network Name",
            allowed_key_mgmt: [:wpa_psk, :wpa_psk_sha256, :sae]
          }
        ]
      },
      ipv4: %{method: :dhcp}
    }

:ok
```

What bothers me is `Connection: :disconnected (44.7 s)`
Any ideas why this happens? I also tried to repower rpi and it does not connect to wifi

fhunleth, 2024-02-19:
What's happening is that your WiFi access point doesn't want protected management frames. The generic WiFi configuration (supports WPA2 and WPA3 if hardware allows) incorrectly required it. If you upgrade VintageNetWiFi or just update to the latest Nerves Livebook, it is fixed (`ieee80211w` parameter should be `1`).

Here's a quick workaround so that you don't need to update:

```
VintageNet.configure("wlan", %{
      type: VintageNetWiFi,
      vintage_net_wifi: %{
        networks: [
          %{
            key_mgmt: :wpa_psk,
            psk: "password",
            ssid: "Network Name"
          }
        ]
      },
      ipv4: %{method: :dhcp}
    })
```

Also, no need to reboot when disconnected. VintageNet will keep trying to connect to WiFi. It won't give up.

arrowcircle, 2024-02-19:
I have eero Pro 6E AP, so not sure if it's good or bad that it is rejected.
I tried switching to wpa_psk as you suggested, it works, but is it safe to fallback to the old standard?
Any other ways to make it work without sacrificing the security?

fhunleth, 2024-02-19:
There might be a setting that you could enable on the access point. It could be called PMF or 80211w.

IMHO, I'm not sure that PMF is a big thing to worry about. I'd be more worried about using WPA2 instead of WPA3. However, you're stuck since the Raspberry Pi doesn't support WPA3 yet. See https://github.com/raspberrypi/linux/issues/4718.