nerves-livebook/nerves_livebook #518: VintageNetWifi.quick_configure broken for Grisp 2

tonnenpinguin, 2024-03-16:
I'm not sure if this is an issue for `nerves_system_grisp_2`, `vintage_net_wifi` or just a docs issue for this repo.

While following the setup guide for the grisp 2 from the readme I can `VintageNetWifi.quick_configure`. However, I kept getting these warnings and errors in the console:

```
13:55:25.536 [error] wpa_supplicant: Process exited with status 255
13:55:25.540 [error] GenServer #PID<0.3828.0> terminating
** (stop) :error_exit_status
Last message: {#Port<0.47>, {:exit_status, 255}}
13:55:26.621 [error] wpa_supplicant: Process exited with status 255
13:55:26.622 [error] GenServer #PID<0.3832.0> terminating
** (stop) :error_exit_status
Last message: {#Port<0.48>, {:exit_status, 255}}
13:55:27.731 [error] wpa_supplicant: Process exited with status 255
13:55:27.732 [error] GenServer #PID<0.3836.0> terminating
** (stop) :error_exit_status
Last message: {#Port<0.49>, {:exit_status, 255}}
13:55:27.771 [warn] RouteManager: new set_connection_status wlan0 -> :disconnected (ifdown)
13:55:27.793 [warn] RouteManager: new set_connection_status wlan0 -> :disconnected (Elixir.VintageNet.Interface.start_configuring/3([file: ~c"lib/vintage_net/interface.ex", line: 669]))
```

Enabling debug logs proved to be more helpful:
```
13:58:41.280 [debug] wpa_supplicant: Line 6: invalid key_mgmt 'SAE'
13:58:41.289 [debug] wpa_supplicant: Line 6: failed to parse key_mgmt 'WPA-PSK WPA-PSK-SHA256 SAE'.
13:58:41.291 [debug] wpa_supplicant: Line 11: failed to parse network block.
13:58:41.292 [debug] wpa_supplicant: Failed to read or parse configuration '/tmp/vintage_net/wpa_supplicant.conf.wlan0'.
```

Seems like 
wpa_supplicant isn't happy with the config created by quick_configure
```elixir
%{
  type: VintageNetWiFi,
  vintage_net_wifi: %{
    networks: [
      %{
        mode: :infrastructure,
        psk: "xxxx",
        sae_password: "VERYSECRET",
        ssid: "MY_SSID",
        ieee80211w: 1,
        key_mgmt: :wpa_psk,
        allowed_key_mgmt: [:wpa_psk, :wpa_psk_sha256, :sae]
      }
    ]
  },
  ipv4: %{method: :dhcp}
}
```

After calling `VintageNet.configure/2` and building the config manually things work fine now, but I think we should probably at least add a hint to the Readme?

fhunleth, 2024-03-17:
Thanks for the report. I have an idea on how to fix and will check next week with other systems updates.

fhunleth, 2024-04-07:
So sorry for the long wait. It turned out that there were multiple issues and the WiFi module on GRiSP2 really, really doesn't like anything that even looks like WPA3.

The BBB had a different problem with WPA3, so I stopped even trying with `VintageNetWiFi.quick_configure/2`. Commit a73de8830fcab056f9a6b78ada2e14166ce0645a fixes this.

closed 2024-04-07 by fhunleth