nerves-livebook/nerves_livebook #563: tflite example model download does not produce a file

lawik, 2024-09-18:
Removing the fancy finch options for progress reporting it works. I did not investigate the issue further right now.

But I just would not get a file written.

fhunleth, 2024-09-18:
@mnishiguchi Are you able to take a look?

mnishiguchi, 2024-09-19:
The notebook seems working on my rpi4, but maybe we might as well make it simpler so it is more stable. I'll work on it.

[image]

mnishiguchi, 2024-09-19:
@lawik Is this something you did to fix the issue? If that is the case, I'll plan to remove them to make the example more stable.

```diff
  defmodule Utils do
    def download!(source_url, req_options \\ []) do
-     Req.get!(source_url, [finch_request: &finch_request/4] ++ req_options).body
+     Req.get!(source_url, req_options).body
    end

-   defp finch_request(req_request, finch_request, finch_name, finch_options) do
-     acc = Req.Response.new()

-     case Finch.stream(finch_request, finch_name, acc, &handle_finch_stream/2, finch_options) do
-       {:ok, response} -> {req_request, response}
-       {:error, exception} -> {req_request, exception}
-     end
-   end

-   defp handle_finch_stream({:status, status}, response), do: %{response | status: status}

-   defp handle_finch_stream({:headers, headers}, response) do
-     req_headers = headers |> Enum.map(fn {k, v} -> {k, List.wrap(v)} end) |> Map.new()
-     total_size = req_headers |> Map.fetch!("content-length") |> List.first() |> String.to_integer

-     response
-     |> Map.put(:headers, req_headers)
-     |> Map.put(:private, %{total_size: total_size, downloaded_size: 0})
-   end

-   defp handle_finch_stream({:data, data}, response) do
-     new_downloaded_size = response.private.downloaded_size + byte_size(data)
-     ProgressBar.render(new_downloaded_size, response.private.total_size, suffix: :bytes)

-     response
-     |> Map.update!(:body, &(&1 <> data))
-     |> Map.update!(:private, &%{&1 | downloaded_size: new_downloaded_size})
-   end
  end
```

lawik, 2024-09-20:
Yes, that fixed it for me when I had the issue.

PR #564 (2024-09-22): Fix file downloader in tflite example
### Description

fix #563

[image]

### Memo

- The side effect from removing what is removed here is merely the progress bar animation going away, which is no big deal.
- ~[progress_bar] package is not used anywhere else so I am removing it~
- Done adjustment for Kino breaking change at v0.11.0

[progress_bar]: https://hex.pm/packages/progress_bar
closed 2024-09-22 by fhunleth